Snapmaker/Luban #680: Snapmaker 2. CNC useless does not import SVG correctly etc.

Warmanguitars, 2021-01-01:
Told by Snapmaker support to ask for help here. I cannot get the CNC section to work at all correctly. Main issues are as follows.

Imported SVG files are far from correct size when imported to workspace, usually a lot smaller.

If the image imported is comprised of say 3 elements and I need to apply different tool paths to each element ( for example cut inside one line and outside one other line and on the third line ) it will not do this. Each, wrongly sized, element of the drawing has to be added independently.

Generally "relief" is the only option shown in Processing Mode. there is nothing else selectable.

I am only trying to cut out basic shapes from flat surface stock and the program does not seem able to do this. There are no instructions in any of the documentation I can find nor any threads that seem to have solved this.

jane-rose, 2021-01-06:
Size problem is same as #502 .
As for the individual toolpaths, we are working on it.
For the mode, we haved add 'vector' mode for svg, look forward to our next version update!
As for instructions, u may find it on https://support.snapmaker.com/hc/en-us.

hotchips, 2021-10-31:
Was having same issue - Worked when I loaded original SVG into illustrator and re-saved with SVG 1.0 spec

closed 2023-03-21 by sumibin