Snapmaker/Luban #688: Camera Picture Alignment

skvarel, 2021-01-08:
The alignment for camera captur doesn't work. No 'Reference Position' possible. I want to align the picture to the middle of the bed.

[image]

jane-rose, 2021-01-11:
Sorry, i don't get your point. It is just on middle of the bed. Did u try to engrave something?

paauserrat, 2021-01-31:
It also happened to me. The image captured apears in a corner and it's not intuitive. It should appear centred in the middle of the bed.

jlaughter, 2021-02-16:
Pretty sure this is simply another request to hide the negative x and y quadrants.  Only the top right quadrant that represents usable bed space should be visible, unless a box is checked somewhere to show them, or unless you're viewing toolpath that goes into -x,-y space.

n1djs, 2021-02-16:
You can drag the picture to the center coordinates as you require.

paauserrat, 2021-02-16:
> You can drag the picture to the center coordinates as you require.

I tried it and It's not possible for me. Using last version.

n1djs, 2021-02-16:
DId you try dragging it, while in edit mode at the top center?

jlaughter, 2021-02-17:
I just did a camera capture and was able to drag the picture around in edit mode - and could even rotate it, which is weird.  But, when moving into process, it's still top right with the work origin set at 0,0.  It appears that the work origin is forced to 0,0 in camera capture - am I interpreting that correctly?  And if so, what value does moving the picture around in "Edit" have?

n1djs, 2021-02-17:
Im only guessing, but I suspect the implementation as currently presented, it just for you to reference that the size of your engraving fits on the size of the object captured by the camera.

Given that snap maker seems to want to operate in a +X/+Y plane, that’s where the camera capture comes from.
The options therefor (as currently implemented) are to reference your engraving coordinates where the snap maker camera operates or drag the capture to the coordinates that your engraving is using. Its been easiest for me just to drag the captured picture to the same coordinates as the engraving.

jeff

jlaughter, 2021-02-17:
The snapmaker operates off work origin as set bu the user - BUT for camera capture I suppose it's critical for the work origin to be fixed to ensure alignment between the captured workspace and image.  So they force the work origin to machine zero/home - which is the bottom left corner of the bed.  That part makes sense to me now.  

Although, from a usability perspective, it would likely be more intuitive if the default work origin for camera capture was the middle of the bed.  I see lots of similar topics where this has confused users, like it confused me.

I'd like to see that change, plus improvements to the camera calibration to account for skew/spherical lens distortion/whatever else that is preventing good engraving accuracy against the image.  Camera capture has the potential to enable a really user and beginner-friendly workflow, but the spatial issues are holding it back.

One last thing:  I see lots of issues where either auto calibration or the user has sets unrealistic camera calibration corner values, and that causes camera capture to fail entirely.  Might want to default those to a reasonable value and provide a "reset defaults" button, or constrain those where neither the machine nor the user can set those corners too crazy.