Snapmaker/Luban #717: Empty material definition causes internal server error and renders Luban unusable

uniederer, 2021-01-20:
After updating to Luban 3.13.1 on my Windows 10 machine, I was playing around with the new material and quality management feature. After closing Luban and reopening it, I found the material and quality setting panes in this state:
[image]

The dev tools reported return code 500 (internal server error) for the following API call:
http://127.0.0.1:63432/api/printingMaterialDefinitions?_=1611179063211

Reverting to an earlier release did show the same behaviour. 

In my despair, I started poking around to find the server's log and came across the `%APPDATA%\Snapmaker-luban\Config` folder and noticed an empty material file `material.48292085.def.json`. I deleted that empty file which made Luban functioning again.

Even though I do not know what caused Luban to create an empty material file, I strongly suggest to make Luban robust enough to be able to deal with malformed configurations. - You know: this "never trust program inputs"-thing.

jane-rose, 2021-01-21:
Delete 'luban' config directory and restart the 'luban' will solve the problem.
On macOs:
`cd ~/Library/Application\ Support/snapmaker-luban`
`rm -rf Config/`
On Windows, deleted the folder `C:\Users\[Your Windows User Profile]\AppData\Roaming\snapmaker-luban`(usually)
If u have custom profiles, before delete the folder, copy the file like `material.xxxxxxx.def.json` or `quality.xxxxxxx.def.json` inside the folder. Then after having restarted luban, paste them into new Config folder will make it work.

uniederer, 2021-01-21:
@jane-rose 
As I wrote above, there is no need to kill the entire Config directory. Just look for empty files (files of zero length) first and remove those. Of course if one of the files is non-empty but corrupted, it might be the fastest solution to simply kill the Config folder. However that's IMHO a bit like using a sledge hammer to crack a nut...

jane-rose, 2021-01-21:
> @jane-rose
> As I wrote above, there is no need to kill the entire Config directory. Just look for empty files (files of zero length) first and remove those. Of course if one of the files is non-empty [...]

Ok, by the way, can i konw what version did you use before upgrading?

uniederer, 2021-01-22:
The previous release was v3.12.3.

closed 2022-12-22 by parachvte