Snapmaker/Luban #722: [Bug][UI] G-code Inspect - Days not shown

kb1ibt, 2021-01-21:
Luban v3.12.3
macOS 11.1 (M1 based MacMini)
When you have a long print job that takes multiple days to print, the Remaining Time and Elapsed Time in the G-code Inspect do not show days, only hours minutes and seconds. As you can see with the attached screen shot from a finished job, it started 1/18/2021 at 9:22pm and finished 1/21/2021 at 8:19am, but yet it shows elapsed time of only 10h56m, instead of the 2d10h56m that is actually took.
[image]

kb1ibt, 2021-01-22:
The source of this problem is [line 37](https://github.com/Snapmaker/Luban/blob/a8b3cf162d6f7afc72dad288b407f51f8c70c787/src/app/widgets/GCode/GCode.jsx#L37) and [line 45](https://github.com/Snapmaker/Luban/blob/a8b3cf162d6f7afc72dad288b407f51f8c70c787/src/app/widgets/GCode/GCode.jsx#L45) of src/app/widgets/GCode/GCode.jsx which both are
`return moment(d._data).format('HH:mm:ss');`

kb1ibt, 2021-01-22:
for `formatElapsedTime` I would suggest something like
```javascript
if (elapsedTime > 86400000){
    return moment(d._data).format('DD[d] HH:mm:ss');
}else{
    return moment(d._data).format('HH:mm:ss');
}
```
Unless I am misunderstanding moment.js, that will allow a job of up to 31 days (which should be sufficient).

jane-rose, 2021-01-22:
Ok, we will follow this up.

ReneMuetti, 2021-02-10:
Under Windows x64 (with WiFi), this display remains completely blank.
After uploading the file to Snapmaker2, Luban disconnects.
If you re-establish the connection manually, the display in the window still remains empty.

kb1ibt, 2021-02-13:
> Under Windows x64 (with WiFi), this display remains completely blank.
> After uploading the file to Snapmaker2, Luban disconnects.
> If you re-establish the connection manually, the display in the window still remains empty.

From what I have gathered from other threads this is actually a completely different issue. The only way to show status in this box, SM2 needs to be connected and started via USB/serial; because WiFi doesn't actually return the current status and location.

ReneMuetti, 2021-02-14:
That's not so nice. One of the selling points for me was that it works without cables (via WIFI).
It would be very nice if you could implement that.
The siplay on the device is also difficult to reach due to the setup position here. WIFI is almost my only option here.

closed 2023-03-21 by sumibin