Snapmaker/Luban #724: Bug in Toolpath

ThM-58, 2021-01-24:
Hello everybody,
I noticed the following error in luban 3.13.1.
I suspect the problem is in the toolpath.
Module laser:
- make a circle
-set the size by entering numbers in the fields W = 10 H = 10
-position the circle x = 5 y = 5
-press on process
!! Position and size are incorrect!
!! the "G-Code" is also generated with the wrong values !!
If I go back to edit and drag the size from the circle with the mouse and set the position with the mouse, then the output in the toolpath looks good ...
If I can send a video via upload, I will do so in the attachment!
MfG ThM

https://user-images.githubusercontent.com/77926210/105632732-1a376080-5e55-11eb-8191-ad2f897e2c95.mp4

jane-rose, 2021-01-25:
Sorry, I cannot reproduce this situation, nor can I play the video attached.

ThM-58, 2021-01-25:
Hello again,
I formulated the process more precisely above!
In the meantime I have installed Luban 3.13.1 under Windows x64 and can reproduce the error here too!
I am now sending the video again as "mov". If necessary, I will convert it to another format I want.
Many thanks for the effort!
MfG ThM

"Snapmaker Luban 3.13.1" Identical error:
"iMac Catalina x64 i7"
"Windows10 x86 i7"

https://user-images.githubusercontent.com/77926210/105702466-1f0b1b80-5f0c-11eb-93ea-210014fb28f7.mov

Neordian, 2021-01-25:
I can confirm a resize bug related to laster paths: Sometimes when i just make some rectangles and duplicate them, some are made bigger and some smaller. But so far I am not sure how to replicate it....

ThM-58, 2021-01-26:
In my opinion, this is what happens:
If I use the keyboard to change the size in the edit field, the width and height are adopted and displayed in the editor window, but not transferred to the toolpath!

It is best understood as follows:
-draw a big circle (100x100)
- change the size using the keyboard in the fields W and H (10x10)
** if I click on process now, everything is still OK !!!
-back to editor window
- move the circle to another position with the mouse
** when I click on process now, the circle appears in the original size (100x100) and not as desired (10x10)
I don't know how to explain better, sorry!
MfG ThM

Neordian, 2021-01-26:
Short remark: It seems to be the same bug as reported in #502 and and #632
At least for my case it seems so.

jane-rose, 2021-01-28:
> In my opinion, this is what happens:
> If I use the keyboard to change the size in the edit field, the width and height are adopted and displayed in the editor window, but not transferred to the [...]

Yes, i can reproduce it now.

closed 2023-03-21 by sumibin