Snapmaker/Luban #726: File size issue (gcode > around 250Mb)

MlvPerry, 2021-01-26:
Hello, I have an issue with gcode files (for 3d printing) larger than approximately 250Mb.
They do not finish transfering to the Snapmaker 2.0 A350, or they crash the display software before starting the print.
I tried with different slicers and different files. For some prints is not possible to further reduce the file size via resolution decrease or print settings optimization...
I see other users have the same or very similar problems, for example [here](https://forum.snapmaker.com/t/file-transfer-issues/8618) and [here](https://forum.snapmaker.com/t/snapmaker-2-0-cura-profile/7537).

scotthsl, 2021-01-26:
@ITmaze Yes, you are right, I am sorry for closing this issue directly before.
I asked my colleague to help to transfer it to luban here.

MlvPerry, 2021-03-01:
Please note that the issue is not caused by the slicing software (or, to be precise, all of the slicing softwares I tried cause this same issue). So there is a problem with the Snapmaker software (firmware).