Snapmaker/Luban #728: Luban hangs when STL is large

dpeart, 2021-01-27:
I'm trying to print large buildings using the A350.  Some of the STL files are 200+MB in size and when I slice them they take over 6GB of memory in Luban.  When this happens, Luban hangs instead of starting the cura slicer process.  Meaning I cannot slice large models for printing on my large printer.

Oddly I can slice the models with Cura and they take 25% of the memory Luban takes.  I prefer Luban for my Snapmaker printer.

Havenissens, 2021-01-30:
I have the same problem when reaching 250m my luban crasches, when tying to slice.

jane-rose, 2021-02-01:
Sorry, it is in progress, but it will not be released soon(have to be full tested).

parachvte, 2021-02-03:
https://github.com/Snapmaker/Luban/pull/737 improves this. (v3.15)