Snapmaker/Luban #732: Switching from Edit to Process causes images to disappear

HED-rhodson, 2021-01-29:
Version 3.13.1
In Laser, I have 4 images - 2 outlines (.svg) for one laser setting (cut), and 2 (.jpg) for engraving.
I can place the images down in Edit mode, and line them up ok, when I switch to process mode, the two images I added using duplicate, disappear from the screen. The first two are fine

The two images are exact copies of each other, and have I matched the settings between the svg, and jpg pairs to be the same.
Both are using Vector - they are line drawings 
This appears to be a new bug, as I don't recall older version of the software behaving this way.

HED-rhodson, 2021-01-29:
I believe the cause is due to SVG file having an ID field in the path 
[image]
A PNG with this field fails, but by manually removing it, allows it to work

HED-rhodson, 2021-01-29:
This one works - the only change was removing ID
[image]

HED-rhodson, 2021-01-29:
I believe the SVG issue is related to an id field being inserted into the svg - as removing it corrects the behavior
[image]

HED-rhodson, 2021-01-29:
This works:
[image]

jane-rose, 2021-02-03:
Thank you for reaching out to us.
- Please send me those images (.svg&.jpg) to me so that we can reproduce this issue at our end.

HED-rhodson, 2021-02-03:
This file ought to work, just import into Lasert->Edit then Switch to Process
[Dragon laser cut 3 mm Head 1-unmodified.zip](https://github.com/Snapmaker/Luban/files/5919087/Dragon.laser.cut.3.mm.Head.1-unmodified.zip)

jenkins007, 2022-01-17:
I also have seen this. On large pictures, it seems to work, but smaller ones disappear.

sumibin, 2023-03-21:
This issue should be fixed now.

closed 2023-03-21 by sumibin