Snapmaker/Luban #735: Help: Set Up and Run Developer

kpninja12, 2021-01-31:
_** This was also posted on the forum. Not sure where the best place is **_

I am a software developer who is interested in contributing to Snapmaker Luban. I purchased the Snapermaker 2.0 and love what it can do but could see several improvements in Luban. I’d love to contribute ideas and even development work to see the success of Snapmaker and Luban grow.

That said, I’m having a difficult time getting Snapmaker Luban set up and ready for development. The Readme guide on GitHub is not very detailed. I set up npm, gulp (which it doesn’t even mention), and node.js. I cloned the repo to my local computer. I started the build process and it told me I needed python (which also isn’t mentioned in the readme). Finally, after getting python installed and running gulp default again, it errors out saying I need Visual Studio. I have Visual Studio Code (VS Code, a lighter version of Visual Studio), can Luban not be developed on VS Code?

A more in-depth guide to setting up a development environment, and listing out the major packages/installations that are required, would be very helpful. I’d love to contribute to the community.

Please help!

kpninja12, 2021-01-31:
[Link to forum post](https://forum.snapmaker.com/t/help-set-up-and-run-developer/15226)

parachvte, 2021-02-04:
I added a “PR-welcome” label here. Since we may have different developer environment, I haven’t meet issues like need to install python (My OS has python pre-installed and I use pyenv to manage several python versions on my local machine). And VSCode is NOT a must to develop Luban, we use different IDEs like VSCode / Atom / WebStorm, of course you can have your own choice.

If you have any problems when setting up the project, feel free to ask / open a PR improving it. We will update document periodically as well.