Snapmaker/Luban #742: [Luban BUG] When resizing camera capture background image luban collapses.

paauserrat, 2021-02-03:
Hi, as the title says, everytime I try to resize the photo processed by the camera using the laser, luban freezes for a long period of time and the work area isn't accessible. Also, the background photo is impossible to resize.

paauserrat, 2021-02-04:
In addition, the background image appears at the top right corner like if this was the full bed. That's not true. It should be fiexd, right? Because if I try to engrave something there, the machine doesn't go there.
[image]

jane-rose, 2021-02-07:
Sorry, for the camera capture, the background image cannot be resized at this time. Plus, which part u try to engrave, the machine can't to go there?

paauserrat, 2021-02-12:
> Sorry, for the camera capture, the background image cannot be resized at this time. Plus, which part u try to engrave, the machine can't to go there?

When I put my design where I want it to go, it doesn't. The machine runs boundary outside the laser bed area. I need to set the home for all axes to the left bottom corner instead of to the center of the bed for it to work.

sumibin, 2023-03-21:
This issue should be fixed now.