Snapmaker/Luban #743: Sometimes Snapmaker disconects from Luban wifi connection when laser camera capture is at half process.

paauserrat, 2021-02-03:
As the tittle says, sometimes I run the camera capture feature of the laser with luban to have the background and see where I want to  engrave. Then, at half process, I get a disconect popup message and my machine disconects from luban but the calibration continues.

jane-rose, 2021-02-21:
This is a wifi connection problem, which version are you using? It should be fixed in luban version 3.13.1 and firmware version 1.11.4.

paauserrat, 2021-02-21:
> This is a wifi connection problem, which version are you using? It should be fixed in luban version 3.13.1 and firmware version 1.11.4.

I'm using the last version of luban (3.13.1) and last firmware version too (1.11.4). Sometimes camera capture fails and disconects. I need to reconect again and sometimes it fals another time or it goes well. Also, my snapmaker sometimes disconects when editing and I need to re-conect. Also, my snapmaker loses my wifi connection everytime I start again the machine.

My wifi connection is good, 1000gb and I'm pritty shure that's not the issue.

sumibin, 2023-03-21:
This issue should currently be fixed. If there is still feedback it will be reopened.

closed 2023-03-21 by sumibin