Snapmaker/Luban #748: Luban laser camera capture calibration four corner point aligment bugs.

paauserrat, 2021-02-04:
First of all, the camera captures the bed with certain inclination. The photo isn't 100% orientated. Here's a photo where this inclination is visible.
[image]

When we try to scroll with the mouse to zoom in and out, sometimes the image moves and the point doesn't show where it was showing. It's not intuitive and it makes this process dificult. This happens too sometimes when we try to move a point to match the square.

jane-rose, 2021-02-07:
As for the performance of zoom operation, yes, this is an optimization point.

sumibin, 2023-03-21:
We have adjusted this part of the interaction. Please submit a new issue if you have further feedback.