Snapmaker/Luban #769: Feature Request: Send to Device via Wi-Fi and Start Print

Eddie-Hartman, 2021-02-13:
With the latest firmware allowing you to reconnect to a previously connected snapmaker machine (#535), it would be great to be able to send gcode via wifi and then tell the machine to begin with print remotely.

I don't mean to print via the play button which requires the computer to remain on during the process, but to send the gcode file to the machine as normal and be able to remotely tell the machine to begin this print.

I use a laptop in a separate room so being able to create the file, send it, and have it start remotely would be a great feature.

redwyre, 2022-01-05:
Didn't Luban used to let you do this? After upgrading to 4 I can't do it anymore.

sumibin, 2023-03-21:
These features have been implemented.

closed 2023-03-21 by sumibin