Snapmaker/Luban #771: Snapmaker Luban 3.13.1 - G-code inspect and Laser engraving process

Mortisaga, 2021-02-17:
I downloaded and installed the version 3.13.1 (Win x64)

Tried to laser engrave as I did many times with v3.10.x - now my G-code inspect remains blank. Only the starting time will be shown.
Second bug (?) I detected: in the former version I had this blue "indication" and a kind of a cursor which showed the process on the image I want to engrave - this is also gone.
I use SM 2.0 A350 via WiFi

Shall I switch back to an older version? I found both of those function pretty handy... I did not test with 3d printing yet.

[image]

Riccardo633, 2021-03-12:
I was running 3.8 until I saw this comment and realised I was way behind, now I'm on 3.13.1 (Win x64) but I've never seen anything in my G-Code inspect on either version - no times, no dimensions. Be keen to hear how I can get the information appearing in there.

jane-rose, 2021-03-12:
@Riccardo633 @Mortisaga  Sorry for the late replay. I have checked it, this part information will only be displayed after starting the printing job.

Riccardo633, 2021-03-12:
Hi @jane-rose thank you for the reply. I find that even after starting the print job I get no information here - attached image is what I see now, mid print. Is there some step I need to complete to enable the display or something?
[image]

jane-rose, 2021-03-12:
3d printing? If is, it will be showed after the nozzle and heated bed are heated. You can try laser/cnc first.
If still happened, you can send us the project file. As for how to export the project file, click the menu at the top one by one, 'File' -> 'save as'.

parachvte, 2021-03-12:
>  I can export a '.snap3dp' file from the '3D Printing G-code Generator' window. I tried to attach that file but it seems GitHub doesn't support that file type. Was that what you meant? I also tried to attach the '.gcode' file with the same result.

@Riccardo633 Compress your project file to upload.

Could you confirm the bug on v3.14.0-alpha version too ? (download [here](https://github.com/Snapmaker/Luban/releases)).

Riccardo633, 2021-03-12:
[HGPExp.zip](https://github.com/Snapmaker/Luban/files/6127693/HGPExp.zip)
I should have thought of that...

downloaded 3.14.0-alpha.12 (win-x64) and still can't see any data in G-code Inspect (although the printer is still part way through the current print job, which is presumably what it would report on).

Thanks!

parachvte, 2021-03-12:
With version (v3.14.0-alpha.12, macOS), 3D Printing. I cannot reproduce your issue.

Job started (heating the bed):
[image]

Job finished:
[image]

Riccardo633, 2021-03-12:
Hmm, then it might be something particular about my installation, but I have now installed three versions and seen the problem in all three. I had also assumed that the Min and Max columns were designed to show the extents of the job but it looks like I had misunderstood that. Regardless, I will keep trying to see if I can at least get the times to show up. Thanks again!

Mortisaga, 2021-03-12:
Hi, I saw this problem only while lasering. As long as I use vector graphics, I can see sent, received, start and elapsed time - no remaining time (finish time appears when the job is finished) ... as long as the program does not lose connection to the printer - after reconnecting everything is blank.
When I print greyscales I see nothing - my screenshot was taken while the machine was working.

I will test 3.14.0 alpha and give a feedback here asap

Riccardo633, 2021-03-15:
>  ... as long as the program does not lose connection to the printer - after reconnecting everything is blank.

That is probably my problem then - my program loses connection every time I start a job. So If I can just fix that then maybe it will work better. I start jobs by sending the file to the machine and then going to the machine and pressing start on the machine interface. At that point I invariably lose connection. 

I did notice the play button that has now appeared in my interface, and looking through other comments it seems that I can start a job using that, so I tried it and it does indeed get time displayed, so I could just swap to using that method. It looks a bit like that requires the computer to remain awake and connected to keep printing though, is that correct? That isn't ideal as my connection does frequently drop out anyway.

Thanks!

Mortisaga, 2021-03-17:
@Riccardo633 
I think you don't need to worry regarding the computer remaining awake.

Should it go to sleep or lose the connection the print will continue - in 3.13.1 I get a message on the printer display, that the connection is lost - IF you confirm this, it will stop the print (I never tested if I can continue it with Luban anyways) - if you do not confirm it will print on and you can always connect again in Luban.

sumibin, 2023-03-21:
In the new version, the job status includes this part of information, and although there are currently some problems due to missing screen returns, it will be fully restored after subsequent screen updates.

closed 2023-03-21 by sumibin