Snapmaker/Luban #779: Undo doesn't do anything?

Galindorf, 2021-02-21:
I am new to this program so I am not sure how much is user error, and how much is program error. I am having lots of difficulties with Luban, but one thing I noticed is that the Undo doesn't seem to do anything, at all. If I delete something by accident and choose "undo" nothing happens. If I move an object by accident, undo does not fix it. Am I the only one? 

Luban 3.13.1
Windows 10 on SurfaceBook original

Wynand-S, 2021-02-24:
I concur - accidentally delete the wrong model, can't undo.
Move the model but need it back exactly on the previous spot, can't undo the move.

jane-rose, 2021-02-24:
Since we can't reproduce it ? Could you please tell me more information of it so that we can troubleshoot it? How about the version and platform?

closed 2023-03-21 by sumibin