Snapmaker/Luban #786: Luban laser process does process to SVG wrongly

paauserrat, 2021-02-22:
I have done a svg with illustrator. When importedto the luban edit software everything's correct. Once I click process, the svg is displayed wrongly and some parts dissapear.

Uploading svg:
[image]
The svg once we process it (wrongly):
[image]

jane-rose, 2021-04-08:
Known  bug, it will be fixed on version v3.15.0.

awesomeruss, 2021-05-04:
I think I have the same issue with 3.15.0. SVG is pretty simple with no groups. Everything is fine until I click 'Preview' at which point horizontal size scales up. 
[image]
[image]
[image]

jane-rose, 2021-05-07:
@awesomeruss Ok, could u send us the project file or svg file to see what happened on these file?

jane-rose, 2021-05-07:
@paauserrat Is this problem solved on v3.15.0? If not, you can send us the svg file too.

awesomeruss, 2021-05-09:
> @paauserrat Is this problem solved on v3.15.0? If not, you can send us the svg file too.

Never mind - I found the problem. My SVG was at fault - the width and height were the wrong way round(!)

Galindorf, 2021-05-10:
I also just had this issue. I have created a grid pattern to test my X axis by making long rectangles. Luban will only path half of them, and they are misaligned also! 

[image]
[image]

Galindorf, 2021-05-10:
Note: I converted the paths from "rectangles" to "Curves" (CorelDraw X8) and then used "Combine" to make them a single curve, then saved as SVG. They showed up and engraved properly after that.

sumibin, 2023-03-21:
Given that there is no available file in the feedback, I can't confirm if the problem is fixed. If the problem persists in the new version please reopen it.

closed 2023-03-21 by sumibin