Snapmaker/Luban #788: Wifi disconnects on every Snapmaker power off

paauserrat, 2021-02-22:
My Snapmaker started to lose the wifi network on every power off after the new update. I'm running Luban last version and Snapmaker firmware last version too.

Every time I open my Snapmaker I need to go to config and type the pasword of the wifi. Then, the machine is connected to wifi forever until I power off the machine and power it on again.

toffentoffen, 2021-02-22:
The very same is happening to me.

sublime24, 2021-02-28:
I have the same issue

zephraph, 2021-03-07:
While I have this same issue, I don't necessarily think it's a Luban problem (so this probably isn't the right place for it). It's reported in the forums here: https://forum.snapmaker.com/t/wifi-not-connecting-at-startup/16628. It started happening after the Feb V1.11.4 update

ncurley999, 2021-03-26:
Just adding my experience here as well, I've also been having this issue this since the Feb update.

genemyers, 2021-04-22:
> Just adding my experience here as well, I've also been having this issue this since the Feb update.

Same here

parachvte, 2021-04-22:
> At first I thought this wasn't the right place for it, but it was likely a regression introduced by #714.

If it happens on Snapmaker reboots, it's not an issue of Luban (though Luban has its own problem on user experience).

We are unable to reproduce the issue in our office, but we have some guesses on what the problem is.
Now we are adding some test code to our firmware, and then build a test firmware (I will upload the firmware later).
Could you help us to use the test firmware to see if the issue still occurs?

btw, we start to use Github project to manage issues posted here, you can check on [project](https://github.com/Snapmaker/Luban/projects/3) to track their status.

paauserrat, 2021-04-23:
Hi! I installed the test firware and it seems to be fixed. At first, I needed to turn wifi off, turn it on again and it automatically reconnected. I turned Snapmaker 2.0 off and on and it automatically reconected to my wifi! Congrats :)

Do I need to re-install the official last firmware release or I can be running the Test you provided without any problem?

zephraph, 2021-04-23:
Thanks for the clarification @parachvte. I'll try it out as well.

parachvte, 2021-04-25:
> Do I need to re-install the official last firmware release or I can be running the Test you provided without any problem?

You can use the test firmware without problems, it's based on v1.12.1 (under testing) to be released.

shuwn, 2021-04-30:
> **新了2021.04.29**：第二次嘗試測試固件，修復了側面情況
> [Snapmaker2_V255.0.0_20210429.bin.zip](https://s3-us-west-2.amazonaws.com/snapmaker.com/download/firmware/Snapmaker2_V255.0.0_20210423.bin.zip)

This update didn't solve the issue.

slipperysam1984, 2021-05-12:
This update didn't solve my issue either.

Sam

sublime24, 2021-05-17:
This update did not solve me issue.

voidrin, 2021-05-27:
Hi, we are still working on this issue. We got some report that the test firmware had solved the problem after reboot the machine, but we still collecting feedback to make sure our changes can fix this issue. 

Please check your machine turn off and turn it on after test firmware upgraded. If the issue still occurs, please let us know.

voidrin, 2021-05-27:
@shuwn @slipperysam1984 @sublime24 
Hi! Did the issue still persist after machine reboot? If you can export the log from the machine, we can check it and try to find out what's the status with the test firmware.

You can export log file from the touchscreen by these steps: 
- Make your U-disk is connected in touchscreen
- Click Settings > Export log
- Wait for touchscreen export log to the U-disk successfully

shuwn, 2021-05-28:
> @shuwn @ slipperysam1984 @ sublime24
> 嗨！重新啟動計算機後問題仍然存在嗎？如果您可以從計算機中導出日誌，我們可以對其進行檢查並嘗試找出測試固件的狀態。
> 
> 您可以通過以下步驟從觸摸屏導出日誌文件：
> 
> * 使您的U盤通過觸摸屏連接
> * 單擊設置>導出日誌
> * 等待觸摸屏導出日誌到U盤成功

FYI

[SC_3.log](https://github.com/Snapmaker/Luban/files/6557462/SC_3.log)

jbrescher1, 2021-06-09:
Having this issue also, newest firmware has not fixed issue.

NiteCrwlr, 2021-06-10:
I'm having the same problem. A350, no wifi connection after machine reboot. Latest version Snapmaker2_V1.12.0.

paauserrat, 2021-06-10:
I'm also having the same problem. It seems fixed at first but it isnt't

omanuke, 2021-06-19:
I encountered this issue but why this silly bug doesn't be fixed immediately?

tom-mat, 2021-07-04:
same problem here, persistent since >6 months... this is ridiculous!

jane-rose, 2021-07-05:
> same problem here, persistent since >6 months... this is ridiculous!

https://github.com/Snapmaker/Luban/issues/788#issuecomment-824811503. Maybe you can try this firmware.

tom-mat, 2021-07-11:
after installing the latest firmware, going back to factory settings via developer mode, installing the firmware her the issue seems solved. I really hopes it stays that way.

artemician, 2021-07-11:
> after installing the latest firmware, going back to factory settings via developer mode, installing the firmware her the issue seems solved. I really hopes it stays that way. 

For what it's worth (I'm late to this) I have never experienced any problems with my WiFi on my Snapmaker. It connects automatically without fail every single time.

When I got the machine I immediately updated to 1.9, it stayed there until I upgraded to 1.12.1, so an update in between 1.9 and 1.12.1 must've screwed it up and it doesn't get reset with subsequent updates so the machine remembers that bug and keeps it until being factory reset.

qriusGate, 2021-12-28:
I just got my snapmaker and am having this issue as well

sublime24, 2021-12-28:
Have you updated? That seemed to fix the issue.