Snapmaker/Luban #789: Camera capture calibration goes wrong

paauserrat, 2021-02-22:
I do the 4 corner calibration the best I can.

Photo with 4 corner calibration:
[image]

My camera captures the bed this way:
[image]

ldp7, 2021-02-23:
In the first calibration result image, the background image you got is tilted. The screen transfers the image via Bluetooth to Snapmaker Luban, which displays it directly without any processing.

Could you please send me the source image of it so that we can troubleshoot it?
On your computer, check the images in a folder named "Temp", and they are generated by the camera directly. You need to send us the jpg files with file names starting with calibration, and here is an image for your reference:

For Windows OS: C:\Users\admin\AppData\Roaming\snapmaker-luban\Tmp
admin is PC's username

For macOS: /Users/admin/Library/Application\ Support/snapmaker-luban/Tmp
admin is PC's username

ldp7, 2021-02-23:
[image]

jane-rose, 2021-02-23:
As for the second result image, you have to use farmware version below 1.9.0 or waiting for the luban version 3.14.0 to solve the mismatch in the third column. Sorry for that happened.

paauserrat, 2021-02-23:
Hi. I did the calibration again as in that folder I had nothing (could be because I started the machine again).

I have the Snapmaker 2.0 a250

The photos on the "tmp" directory:
[image]
[image]
[image]

If you need them too, I did photos of my calibration proces:

First time I run 4 point calibration the points appear where they want. Also, when I click reset they go to this position.
[image]

Then I align the points:
[image]

The calibration is done:
[image]

The photo displayed as background:
[image]

For instance, if I try to engrave something in a specific position of the background my machine doesn't go there. Why?

jane-rose, 2021-02-24:
Could you send your engraving operation steps and engraving result photos?

lephiloux, 2021-04-02:
The problem for me is that background is not centred on the coordinate system of the grid 
[image]

paauserrat, 2021-04-11:
Please, this is important. I'm running luban 1.14 and my laser doesn't capture the bed correctly.

jane-rose, 2021-04-11:
> Please, this is important. I'm running luban 1.14 and my laser doesn't capture the bed correctly.

The abnormal phenomenon is that the photo on calibration page is tilted, can you send the result of your carving?

ldp7, 2021-04-13:
Please refer to this video and send us those images in the "temp" folder. Thank you in advance.

How to troubleshoot the Camera capture feature if the images are titled:
https://drive.google.com/file/d/1NcqDwbc0LxLScpjfv-MScim7xJxyKJQX/view?usp=sharing

You need to check the images in a folder named "temp", which are generated by the camera directly.
Windows OS: C:\Users\admin\AppData\Roaming\snapmaker-luban\Tmp
admin is PC's Profile
Mac Os: /Users/admin/Library/Application\ Support/snapmaker-luban/Tmp
admin is PC's Profile

ldp7, 2021-04-13:
@paauserrat  

We want to see the calibration image after you click the "Confirm" button in Snapmaker Luban.

paauserrat, 2021-04-17:
> Hi. I did the calibration again as in that folder I had nothing (could be because I started the machine again).
>
> I have the Snapmaker 2.0 a250
>
> The photos on the "tmp" directory:
> [image]
> [image] [...]

I already uploaded them here. Didn't I? Do you want me to do the calibration again and send them? Do you need the image name too?

pauserratgutierrez, 2022-03-06:
Hi, I still have this issue. May I contact sales service?

jane-rose, 2022-03-06:
Hi. You can contact [support service]([support@snapmaker.com](mailto:support@snapmaker.com))(may a remote video to reproduce and solve your problem). The photo displayed as background seems correct. By the way, the  problem is `if u try to engrave something in a specific position of the background your machine doesn't go there.`? And what firmware version and luban version you are using now.

sublime24, 2022-04-03:
Hello, I have the same issue was this ever resolved?

jane-rose, 2022-04-03:
> Hello, I have the same issue was this ever resolved? I went to the path specified and found an old calibration image. I deleted, closed snapmaker luben, reopened and the calibration image reappeared in the tmp file. I do no think that the calibration image is saved locally, but on the machine (a350), I assume it grabs the information from the machine (a350) directly and loads a tmp image on the PC, running snapmaker. Is there a way to clear the snapmaker local cache or specifically the calibration image saved on the snapmaker?

Yep, the `calibration image` is saved on the fabsreen. U can redo the `calibration operation` on fabsreen to change `calibration image`.

closed 2023-03-21 by sumibin