Snapmaker/Luban #803: [BUG] luban does note render gcode afetr slicing when using supports

beunwa, 2021-02-27:
Every times I use  the support option for 3d printing (A350) with complex models the luban crash with the following error :

`2021-02-27T17:17:37.992Z - error uncaughtException: "toString()" failed
Error: "toString()" failed
    at stringSlice (buffer.js:560:43)
    at Buffer.toString (buffer.js:633:10)
    at Object.fs.readFileSync (fs.js:601:41)
    at Object.fs.readFileSync (ELECTRON_ASAR.js:538:29)
    at /opt/Snapmaker Luban/resources/app/server/index.js:1:276251
    at ChildProcess.<anonymous> (/opt/Snapmaker Luban/resources/app/server/index.js:1:277158)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:554:12)
2021-02-27T17:17:37.993Z - error: uncaughtException: "toString()" failed date=Sat Feb 27 2021 18:17:37 GMT+0100 (CET), pid=13537, uid=1000, gid=1000, cwd=/opt/Snapmaker Luban/resources/app/server, execPath=/opt/Snapmaker Luban/snapmaker-luban, version=v8.9.3, argv=[/opt/Snapmaker Luban/snapmaker-luban], rss=1078898688, heapTotal=59064320, heapUsed=52035640, external=870777666, loadavg=[6.78857421875, 2.9453125, 1.5439453125], uptime=4159, trace=[column=43, file=buffer.js, function=stringSlice, line=560, method=null, native=false, column=10, file=buffer.js, function=Buffer.toString, line=633, method=toString, native=false, column=41, file=fs.js, function=Object.fs.readFileSync, line=601, method=readFileSync, native=false, column=29, file=ELECTRON_ASAR.js, function=Object.fs.readFileSync, line=538, method=readFileSync, native=false, column=276251, file=/opt/Snapmaker Luban/resources/app/server/index.js, function=null, line=1, method=null, native=false, column=277158, file=/opt/Snapmaker Luban/resources/app/server/index.js, function=null, line=1, method=null, native=false, column=13, file=events.js, function=emitTwo, line=126, method=null, native=false, column=7, file=events.js, function=ChildProcess.emit, line=214, method=emit, native=false, column=16, file=internal/child_process.js, function=maybeClose, line=925, method=null, native=false, column=11, file=internal/child_process.js, function=Socket.stream.socket.on, line=346, method=on, native=false, column=13, file=events.js, function=emitOne, line=116, method=null, native=false, column=7, file=events.js, function=Socket.emit, line=211, method=emit, native=false, column=12, file=net.js, function=Pipe._handle.close [as _onclose], line=554, method=close [as _onclose], native=false], stack=[Error: "toString()" failed,     at stringSlice (buffer.js:560:43),     at Buffer.toString (buffer.js:633:10),     at Object.fs.readFileSync (fs.js:601:41),     at Object.fs.readFileSync (ELECTRON_ASAR.js:538:29),     at /opt/Snapmaker Luban/resources/app/server/index.js:1:276251,     at ChildProcess.<anonymous> (/opt/Snapmaker Luban/resources/app/server/index.js:1:277158),     at emitTwo (events.js:126:13),     at ChildProcess.emit (events.js:214:7),     at maybeClose (internal/child_process.js:925:16),     at Socket.stream.socket.on (internal/child_process.js:346:11),     at emitOne (events.js:116:13),     at Socket.emit (events.js:211:7),     at Pipe._handle.close [as _onclose] (net.js:554:12)]
`

beunwa, 2021-02-27:
precision : this happen when using full size capabilities of the A350

jane-rose, 2021-03-08:
Sorry for the late reply. Could you provide your project file so that we can reproduce it.
As for how to export the project file, at 3dp tab, click the menu at the top one by one, 'File' -> 'save as'.

closed 2023-03-21 by sumibin