Snapmaker/Luban #806: Bug in CNC path

marsiancba, 2021-03-01:
This is a problem I encouter quite often when generating CNC toolpath:

[image]

Seems that sometimes when tool goes over a ridge, it "forgets" to drop down to correct depth after the ridge.

(Luban 3.13.1)

jane-rose, 2021-03-08:
Could you provide your project file so that we can understand exactly what your problem describes. 
As for how to export the project file, click the menu at the top one by one, 'File' -> 'save as'.

marsiancba, 2021-03-08:
@jane-rose 

https://www.dropbox.com/s/ndlfdc2tji8u80n/bug_cnc.snapcnc?dl=0

[image]

jane-rose, 2021-03-08:
Ok, we can reproduce it. we will repair it as soon as possible. Thanks for your reply.

parachvte, 2021-04-27:
This issue has been fixed in https://github.com/Snapmaker/Luban/pull/836 and released in Luban versions >= 3.14.0.

closed 2021-04-27 by parachvte