Snapmaker/Luban #850: Error empty filament

baron-digit, 2021-03-10:
I updated my snapmaker 2.0 today top the current Versions dir Windows 10.

I.printed a Job for around 6h directly fein an USB file and around 40 false filament errors. Not sure if something changed with the new firmware, nur quite annoying...

jane-rose, 2021-03-11:
Could you send your error info? Just 'empty filament'. 
One more thing, Could you provide your project file so that we can reproduce the issue.
As for how to export the project file, click the menu at the top one by one, 'File' -> 'save as'.

baron-digit, 2021-03-13:
The last time I started the print directly via USB, but today I tried another file directly from Luban and have the same error (409). It ended up that I wasn't able to restart the print, although loading the filament immediately extruded the filament.... 

it was just a simple STL-file

[funnel.zip](https://github.com/Snapmaker/Luban/files/6134487/funnel.zip)

sumibin, 2023-03-21:
If this problem still exists in the new version, please reopen it.

closed 2023-03-21 by sumibin