Snapmaker/Luban #870: Door Detection On / Off is switched in 3.13.1

amv5012, 2021-03-12:
If I turn on Door Detection, it acts like its off. I made the mistake of thinking the laser wasn't running while Door detection was on. 
and if its Off, its actually on. 
[image]

jane-rose, 2021-03-16:
Could you tell us what version of the firmware you are using?

amv5012, 2021-03-16:
Hi Jane,

This is what I see on my screen, see attached

amv5012, 2021-03-17:
same issue in luban 3.14.0

amv5012, 2021-03-18:
[image]

amv5012, 2021-03-18:
[image]

amv5012, 2021-04-26:
seems to be fixed in 3.15.0

closed 2021-04-26 by amv5012