Snapmaker/Luban #880: Bug: Luban 1.13.1 (and earlier versions) unable to open laser project files

ozzyman, 2021-03-16:
Linux version of Luban does not open saved project files.

# 🐞 bug report

### Affected Version(s)
All Linux versions

### Is this a regression? (optional)

No

### To Reproduce
Steps to reproduce the behavior:
Method 1:
1. In Luban, select Laser Gcode Generator
2. Click on FILE -> RECENT FILES
3. File browser appears with no files listed

Method 2:
1. In Luban, select Laser Gcode Generator
2. Click on FILE -> OPEN FILE
3. File browser appears, browse to location of file
4. No files are listed

Expected behavior

Project files are listed, and can be opened in FILE -> OPEN dialogue

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Ubuntu 20.04
- Printer: Snapmaker A350

</code></pre>

jane-rose, 2021-03-16:
Thank you for filling in the issue patiently.
One more thing, could you provide the project file?

ozzyman, 2021-03-25:
I think I have found the problem. The Linux version does not appear to save the file correctly. The file is missing the .snaplzr extension. e.g the file is saved as "project" instead of "project.snaplzr"

jane-rose, 2021-03-25:
Ok, now we know that, thanks for your reply.

parachvte, 2021-04-27:
Closing this issue. Seems like it's platform specific problem, file save dialog varys in different OS.

closed 2021-04-27 by parachvte