Snapmaker/Luban #887: Feature Request: Add PETG preset

mr-manuel, 2021-03-17:
# 🐞 feature request

I'm missing a preset for material and printing settings für PETG material. I bought a Snapmaker 2.0 and PETG filament from your shop but the PETG presets are missing in Luban.

I found a post in the forum also discussing about it: https://forum.snapmaker.com/t/petg-printing-profile/5726/19

Can you please add such a preset?

WillBeebe, 2021-03-23:
I added the settings from the forum [here](https://github.com/Snapmaker/Luban/pull/893).

jane-rose, 2021-03-24:
Thanks.But our test of petg's printing profile is in progress, I may wait until the test is over before adding it.

WillBeebe, 2021-03-24:
No problem, just trying to help 🙏

parachvte, 2021-08-18:
PETG profile is available in Luban v4.0.0.

closed 2021-08-18 by parachvte