Snapmaker/Luban #899: Bug: Wrong Font in Header of Material & Print Settings Dialogue

lichtwellenreiter, 2021-03-25:
# 🐞 bug report

### Affected Version(s)
 The issue is caused by version 3.14.0

### To Reproduce
Open the Dialog to Customize Materials and Print Settings

Exception or Error (optional)
[image]

<pre><code>
Wrong Font in Header. It is a Font with serifs like times new roman, it should be one without serifes. It is worn in this CSS class:
.views-PrintingManager-styles__manager-type--1lzj2 views-PrintingManager-styles__selected--zbziA
The font currently used is font-family: PingFangSC-Medium, PingFang SC;
Assuming this is a Font to display Chinese Signs? On Latin Computers it dies not work
</code></pre>

Expected behavior
On Latin PC's change font to Appropriate one Like in other UI: 
<pre><code>font-family: Arial, "Helvetica Neue", Helvetica, sans-serif;</code></pre>

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 10 x64 (English US Language Pack)
- Luban: 3.14.0

</code></pre>

spacegaier, 2021-03-25:
I can confirm the issue. Noticed the same thing (same Luban and OS version).

parachvte, 2021-04-27:
This has been fixed in #913

closed 2021-08-18 by parachvte