Snapmaker/Luban #900: Feature Request: Apple Silicon (M1) optimisation

Airmonkdev, 2021-03-26:
Hello,
I've just seen that the last version of Luban (v3.14.0) for macOS is still for Intel based machine.... it works fine on my MacBook Air M1 with Rosetta but could you please build an Universal version?

troian, 2022-09-21:
@parachvte any updates on this? it is over a year since issue started.
lots of people are using M1 machines and experience with Luban via Rosetta is horrible

troian, 2022-09-22:
i have actually built an arm64 release for m1 silicon. seems to work ok

closed 2023-03-21 by sumibin

oizone, 2024-02-04:
I just noticed that the arm64 build doesn't work without rosetta. When opening an stl, it hangs at 75%.

This is in the error log:
`2024-02-04T10:52:41.588Z - error uncaughtException: spawn Unknown system error -86
Error: spawn Unknown system error -86
    at ChildProcess.spawn (node:internal/child_process:415:11)
    at Object.spawn (node:child_process:707:9)
    at Slicer.slice (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:1014398)
    at Slicer.end (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:1015201)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:852808
    at handleCheckModel (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:853114)
    at t.value [as channelMiddleware] (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:194915)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:196160
    at Socket.<anonymous> (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:194347)
    at Socket.emit (node:events:394:28)
    at Socket.emitUntyped (/Applications/Snapmaker Luban.app/Contents/Resources/app/node_modules/socket.io/dist/typed-events.js:69:22)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/node_modules/socket.io/dist/socket.js:614:39
    at processTicksAndRejections (node:internal/process/task_queues:78:11)
2024-02-04T10:52:41.588Z - error app uncaught exception Error: spawn Unknown system error -86
    at ChildProcess.spawn (node:internal/child_process:415:11)
    at Object.spawn (node:child_process:707:9)
    at Slicer.slice (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:1014398)
    at Slicer.end (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:1015201)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:852808
    at handleCheckModel (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:853114)
    at t.value [as channelMiddleware] (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:194915)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:196160
    at Socket.<anonymous> (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:194347)
    at Socket.emit (node:events:394:28)
    at Socket.emitUntyped (/Applications/Snapmaker Luban.app/Contents/Resources/app/node_modules/socket.io/dist/typed-events.js:69:22)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/node_modules/socket.io/dist/socket.js:614:39
    at processTicksAndRejections (node:internal/process/task_queues:78:11) {
  errno: -86,
  code: 'Unknown system error -86',
  syscall: 'spawn'
}`

I assume some plugin requires amd64?