Snapmaker/Luban #912: Bug: Adding more than six text objects fails or takes very long

deimi, 2021-04-06:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 3.14.0

### Is this a regression? (optional)

 Not sure, but i think it was working with 3.13

### To Reproduce
Steps to reproduce the behavior:
When adding more than six text objects to a cnc/laser project, it fails to add the object. See the attached GIF.  
[image]  

I also checked for the api responses with the developer tool. There I can see that I don't get a response for my request of adding a text object.
[image]  

Sometimes the response comes after very long time. In this example after 59sec!!!!
[image]

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Ubuntu 20.04
- Printer: Snapmaker A350

</code></pre>

deimi, 2021-04-06:
This problem also occurs when you not just add more then six text object in a row, but also edit them in between. This means that when I working on a project with several text object or when I delete one and add a new instead, the problem arises as well after about six objects.

deimi, 2021-04-19:
@parachvte @jane-rose any clue/idea?

jane-rose, 2021-04-20:
On my macbook, there is indeed a lag of more than 10 seconds before rendering.

deimi, 2021-04-26:
Same behavior in 3.15.0

parachvte, 2021-04-28:
@deimi 
We were unable to reproduce the behavior on Windows (both versions).
We'll retry on Ubuntu...

closed 2023-01-30 by parachvte