Snapmaker/Luban #920: Feature Request: option to choose between square and circle outlines / boundary

paauserrat, 2021-04-11:
# **🐞 feature request**

**### Is your feature request related to a problem? Please describe.**

I usually engrave things that aren't squares. I have circles and everythime I run boundary I need to "imagine" if everything is okay.

**### Describe the solution you'd like?**

It would be amazing to have an option to run a cercle boundary.
To generate the outline of every task you can:
1. center the point in the exact middle of the task.
2. search for the extreme points of the task.
3. create a circle that matches the extreme points.

jane-rose, 2021-04-21:
It's a good point, we will discuss for it. Any progress will update in this issue. Thanks.

paauserrat, 2021-04-27:
I have another idea. Instead of only being limited to run a boundary shape, is it possible to make the boundary go following the perimeter of whatever the engraving or carving thing is?

parachvte, 2021-04-28:
> make the boundary go following the perimeter of whatever the engraving or carving thing

That's what we thought.
I'll mark the FR as Need triage cause we don't optimize it in the recent quarter.