Snapmaker/Luban #942: Bug: 4 axis CNC gcode generation does not remove central material

SchroedingerHepcat, 2021-04-16:
# 🐞 bug report

### Affected Version(s)
The issue is caused by version 3.14.0

### To Reproduce
Steps to reproduce the behavior:
1. Go to CNC carving workspace
2. Select 4-axis
3. Set model size
4. Add a model that has material to remove centrally that is angularly elipsed, like a horse 
5. Click on 'Process'
6. Click on 'Simulate'
7. See how central material is not removed behind the legs, in spite of being readily accessible from a different angle.

[image]

You can see how the material between the head, and wings is not removed, as well as the material between the legs

### Expected behavior
Material should be removed from within the central area of a model when accessible with the given bit size.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [Manjaro Linux]
- Printer: [Snapmaker 2.0 A350 with enclosure and rotary model]

</code></pre>

jane-rose, 2021-04-21:
Known bug, any progress will update in this issue, thanks.

gysnst2, 2021-04-28:
Still happens in 3.15

parachvte, 2021-04-28:
It's due to the implementation of simulation.
Simulation can be used as a reference, it's not 100% accurate. Toolpath view will be more accurate.

We are going to re-implement 3D model simulation of CNC later.