Snapmaker/Luban #944: Feature Request: set if mill should follow the line on the outside or inside in cnc

EHoglid, 2021-04-18:
When importing a 2 dimension DXF file it should be great if you can decide if the mill should follow the lines on the outside or inside for each line.

For example. If you want to make a hole in a wooden plate, you want to follow the line on the inside, removing the radius of the CNC bits.
If you want to carve out a plate, you want to follow the line on the outside.

jane-rose, 2021-04-21:
It's on the way. At this time, the outline for svg is correct.But for dxf, it has some problems.

EHoglid, 2021-04-21:
[image]

Great, because in this case it is important for me that the mill followed the line on the inner side, keeping the dimension correct of the hole, not the cutout circle.

Erik

HorstBaerbel, 2022-06-05:
Imo this is an essential function. At the moment the option "Outline" is plain wrong for both DXF and SVGs, if I'm not mistaken. It does a double cut, which totally unnecessary.
[image]
[image]

Arkktos, 2023-04-02:
Any update on this one?

lalilaloe, 2023-04-30:
Please fix this asap

Wynand-S, 2023-05-01:
> It's on the way. At this time, the outline for svg is correct.But for dxf, it has some problems.

It's been 2 years,,,

lwalbert, 2025-04-19:
> It's on the way. At this time, the outline for svg is correct.But for dxf, it has some problems.

Still waiting for this inline/outline update.