Snapmaker/Luban #945: Bug: Structure support created in mid air (3D Printing G-Code Generator)

mr-manuel, 2021-04-18:
# 🐞 bug report

The G-Code generator creates a support structure in mid air that is nowhere attached:
[image]

### Affected Version(s)

 The issue is caused by version 3.14.0

### To Reproduce
Steps to reproduce the behavior:
1. Import the STL file [test.zip](https://github.com/Snapmaker/Luban/files/6331765/test.zip)
2. Import profile from the zip or use this Support settings:
[image]
3. Generate G-Code
4. View layer 76, here you can see that the structure is printed in the air and nowhere attached.
 

## Expected behavior

The support should be attached to the rest of the surrounding support or a carrier should be created before the flying support can be placed.

## 🌍  Your Environment

**Platform:**
<pre><code>- Operating System: Windows 10 x64 Version 2004
- Printer: Snapmaker A250
</code></pre>

jane-rose, 2021-04-19:
Sorry, we can't reproduce it. Or your can send the project file to us. Try to rotate or move the model to see if this bug still occurs

mr-manuel, 2021-04-19:
Yes sure. If I rotate the oject there are only flat surfaces, so the error could not appear. Here the project file: 
[test.zip](https://github.com/Snapmaker/Luban/files/6333848/test.zip)

jane-rose, 2021-04-21:
It's a problem with cura. At this time, you may change the support settings like density to aviod this

sumibin, 2023-03-21:
The issue has been fixed.

closed 2023-03-21 by sumibin