Snapmaker/Luban #972: Feature Request: User profiles for Laser work like in 3D printing profiles

paauserrat, 2021-04-23:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.

It would be a timesaver to have profiles in the laser part of luban. Something like in the 3d printing part where we can choose between pla, abs and other configs like FAST print, QUALIRT print... So in the laser we should have profiles with our parameters saves. This would help us save time. If we want to laser engrave a greyscale image which is the same everytime, it would be easy to check the same profile, with the parameters saved for us :)

parachvte, 2021-04-26:
As you can see, we added tool profiles for CNC in Luban v3.15.0.

There are two types of parameters: 
- one type is for tool / material, it relates to certain resource.
- another type is for process, like target depth and jog height.

We have a plan for making the first type to profiles. The second type needs to be adjusted per object per job, it's not yet considered how we deal with it.

Currently we are rewriting Laser and CNC part of image processing and G-code processing in C++, which provides us better performance dealing with large size jobs. Then we can come to the laser profile thing since parameter structures may change during the refactoring. We'll mark the feature request as low priority and bring it back when proper.

sumibin, 2023-03-21:
This feature is currently being released.

closed 2023-03-21 by sumibin