Snapmaker/Luban #975: Latest Linux version laser module is completely borked.

ozzyman, 2021-04-25:
I recently re-installed Ubuntu 20.04.2, and installed the latest Luban (3.14.0) as part of the process. The laser module is completely mangled, and appears to be some strange subset of the CNC menu. 

In laser edit mode, the right hand panel in the laser screen shows "Job type (3 Axis/4 Axis)", "Object list" and "Actions". None of the normal laser controls are present. In process mode, the options are "Preview toolpath", "Generate Gcode", "Load Gcode to workspace" and "Export Gcode". Again. none of the normal laser process options are present. I have uninstalled and reinstalled with no improvement.

ozzyman, 2021-04-25:
Forgot to install libgconf-2-4. Perhaps this should be included in the .deb, as it appears to be needed to run the package, not just to compile it.

closed 2021-04-25 by ozzyman

parachvte, 2021-04-25:
I'm not sure if it should be included in .deb file, it's documented in our README at least, and installed in Travis build environment.

https://github.com/Snapmaker/Luban/pull/457/files