Snapmaker/Luban #976: Feature Request: make the maxium value of machine be the speed limit

paauserrat, 2021-04-25:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.

We shouldn't be able to go over the maxium velocity of each motor. Luban should indicate you if the value you provided is impossible for the machine to make a move because of max velocity.

### Describe the solution you'd like?

Make the velocity imput not able to go over a certain max value of the machine

parachvte, 2021-04-26:
Put onto backlog.

paauserrat, 2021-04-26:
Amazing. Just to inform, it should be added into every velocity input of luban (not only to 3D moves and retraction velocity... but also to CNC and Laser velocities)

sumibin, 2023-03-21:
Given that the maximum speed has been limited in the firmware. (Luban also currently has restrictions on it.)This issue will be closed.

closed 2023-03-21 by sumibin