Snapmaker/Luban #977: Feature Request: 3D printing Retraction & Z Hop improvements

paauserrat, 2021-04-25:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.

When 3D printing sometimes the printer needs to move from one point to another without extruding, and that's why retraction is used. Z Hop is very usefull to move the bed and don't ruin prints. I don't know which method you use to detect if Z Hop needs to happen but sometimes the printer moves from one point to another point without doing a Hop and it makes a small line to the prints. I think it's a problem related to retraction but also to Z Hop.

### Describe the solution you'd like?

Maybe look the code of retraction and Z Hop and see if there are some cases when it would be usefull and isn't activated.

parachvte, 2021-04-26:
There are several parameters that triggers retraction and an option to do Z hop on every retraction.
Have you check and test the parameters?

[image]

paauserrat, 2021-04-26:
Yes, I know that luban has some parameters to activate retraction and Z Hop. I use them but sometimes the machine moves from one point to another without doing a Z Hop (when Z Hop is activated) the nozzle moving makes the surface of the 3D print have small lines.

parachvte, 2021-04-26:
We are not looking to slice optimization now, maybe later this year.

In Luban, we are still using CuraEngine 3.6 for 3D slicing. You can checkout its parameters to see if any is useful and we haven't expose in Luban yet.

sumibin, 2023-03-21:
This is controlled by a number of more advanced parameters. Luban is currently open for modification of these parameters.

closed 2023-03-21 by sumibin