Snapmaker/Luban #978: Luban 3.15.0 - echo:Unknown command: "M1011"

Mortisaga, 2021-04-25:
I installed 3.15.0 - when I connect Luban via USB (Com-port) I get the message "echo:Unknown command: "M1011" ".
It does not appear by connecting via WiFi.

parachvte, 2021-04-26:
Confirmed bug. Should be fixed ASAP.

Mortisaga, 2021-04-26:
Thank you for the fast response. I did not find something in the formum - now I see, this post was created 1h after I posted here. For the devs - is it better to post issues/possible bugs in the snapmaker forum or at github?

parachvte, 2021-04-26:
> For the devs - is it better to post issues/possible bugs in the snapmaker forum or at github?

Both. Feedback on forum will go back to our backlog as well. 
On the forum you will have our kindly technical support collecting and tracking the issues. Especially when you don't know where the issue locates (Luban / firmware / hardware).
On Github you will get direct feedback from devs, track issue status.

parachvte, 2021-08-10:
The issue should be fixed in versions >= 3.15.2

Mortisaga, 2021-12-22:
the problem still remains in 4.1.2 (or it returned), now it is M1012

[image]