Snapmaker/Luban #981: Feature Request: xxxx in laser/cnc

gysnst2, 2021-04-28:
- [ ] 

# 🐞 feature request

### Is your feature request related to a problem? Please describe.

### Describe the solution you'd like?

parachvte, 2021-04-28:
I sort out the requests you asked:

- Put "Door Detection" option to Enclosure screen
- Persist "Door Detection" and "Light" settings, be able to open light automatically
- Add a "Light Splash" when the machine starts

Right? @gysnst2 

For the first request, "Door Detection" option will be removed (or already removed) because it's a security restriction of products. So you won't be able to configure door detection any more, it's turn on by default.

gysnst2, 2021-04-28:
On the door detection it will probably be a firmware issue. 

By the light's do splash on power on but default off.  If the light settings could be set to on or off in the enclosure menu.  And saved permanently that would be a better option.  Some may want the lights to default to on some may want them off.

parachvte, 2021-04-30:
Persist light settings is scheduled on firmware. Luban may follow up when it's done.