Snapmaker/Luban #986: Bug: Button is hidden behind remove background button in Laser

paauserrat, 2021-04-29:
# 🐞 bug report

The version 3.15.0 introduced new things. One of them is a new blue button in the top right corner. In the laser page in luban, this button is hidden by another button, the remove background.

[image]

paauserrat, 2021-04-29:
The hidden button is "Back to object view".

Galindorf, 2021-05-09:
Can confirm: This is really annoying. With the Luban update I could not find where to configure my laser settings... I had no idea the Object View was there and I could not find how to even select an object to make a tool path OF!

jane-rose, 2021-05-10:
OK, we got it.

closed 2022-12-22 by parachvte