Snapmaker/Luban #987: Bug: luban 3.15.0 laser engraving "diagonal2" line direction not working

paauserrat, 2021-04-29:
Hi, I have recently updated to luban 3.15.0 and I noticed that laser line direction "Diagonal2" isn't working and does vertical lines. I don't know if the other parameters are working okay or need a fix too.

This wasn't happening with the last version of luban, it started happening when I updated to luban 3.15.0.

paauserrat, 2021-04-29:
I noticed that "diagonal" isn't working too.

paauserrat, 2021-04-29:
Oh, I noticed that diagonal and other parameters only work for Dot (high quality) and not for Line. Why?

parachvte, 2021-04-30:
Could you provide project file for demonstration?

Diagonal options are working good on my side.

[image]

paauserrat, 2021-04-30:
Sorry, I got confused. I wrote it wrong. Theese parameters are working for Line (Low Quality) but not for Dot (High Quality).

parachvte, 2021-04-30:
Selecting "Dot" the tool head will engrave image dot by dot.

In G-code, it will be like

```
M3 ;Turn on laser
G4 ;Wait several milliseconds
M5 ;Turn off laser
G0 X{} Y{} ;Move to next dot
```

The impact of moving direction is too little to consider. Maybe maybe maybe we will unify their process in the future.
I'd rather closing the issue for now.

paauserrat, 2021-04-30:
Okey. I suggest to remove the option to choose theese parameters using dot mode as this confuses the user!

NomadicVoxel, 2021-05-21:
> Selecting "Dot" the tool head will engrave image dot by dot.
>
> In G-code, it will be like
>
> ```
> M3 ;Turn on laser
> G4 ;Wait several milliseconds
> M5 ;Turn off laser
> G0 X{} Y{} ;Move to next dot
> ```
>
> The [...]

I would **strongly** beg to differ! I'm trying to laser-cut on a piece of wood that keeps getting shaken off because it keeps doing vertical scanlines too fast. Speed wouldn't be as big an issue in horizontal. And it's taken me until finding this post to find out that the line direction doesn't affect dot mode!

paauserrat, 2021-05-21:
> I would **strongly** beg to differ! I'm trying to laser-cut on a piece of wood that keeps getting shaken off because it keeps doing vertical scanlines too fast. Speed wouldn't be as big an issue in [...]

That's true! In diagonal mode speed is divided by 2, so if the speed is 6000 mm/second in horitzontal (which only one axis is moving), it will be 3000 mm/second in diagonal (3000 mm/second for x and 3000 mm/second for y). This will help us eliminate vibration and other things! Please @parachvte consider reading this and changing the way this works!

sumibin, 2023-03-21:
Given that Luban has currently removed the direction parameter. We will consider extending this feature.