Snapmaker/Luban #1017: Suggestion: make bed come forward on completion of file

Galindorf, 2021-06-07:
It would be nice if, after a project is finished, the bed would move itself all the way forward so that the project is easy to reach. It sometimes does this, sometimes doesn't. I'm not sure why. I am short and my Snapmaker A350 is on a cabinet. When my laser cut project is done I have to reach head and shoulders into the enclosure to pull up the tape securing the project and pull it out. 

Thanks!

jane-rose, 2021-06-18:
This feature has been implemented. "It sometimes does this, sometimes doesn't." Could you describe it specifically? Is it related to the toolhead or something else.

Galindorf, 2022-05-02:
Apologies for the late reply. When I cut a project, the toolhead itself homes but the cutting bed stays where it is, meaning I have to manually move it forward with touch screen controls or use Luban to "home" it foward.