Snapmaker/Luban #1038: Feature Request: Pause print at specific heights in 3DP

uniederer, 2021-06-25:
# The problem
There are various cases, where it could be beneficial to pauses the print in order to...
- ...insert objects into the print
- ...change filament (e.g. different color)
- ...place checkpoints to manually inspect the print
- ...

At the moment, I use a stopwatch to approximate the correct time to pause the print. However this is neither precise nor convenient.

# The solution
It would be desirable to have the possibility to provide Luban with a list of heights where I want to have the print paused. After generating the G-Code, Luban shall insert pause commands in print's G-code.

tyeth, 2021-09-21:
I was curious about the pause situation with my old printer but now that I have snapmaker2+1 i am also curious about this topic, I heard M600 should be a pause for manual or automatic filament change. I'm not thinking it's fair to worry about retracting the filament fully etc, more just pause print with a tiny bit of retraction for oozing and move away from the bed / area of print to allow easy change and manual run-through of new filament, with ideally a user interface button to resume...

Ulopes, 2021-10-02:
Hi, I'm also curious about this. I have a SM2-A350 and I need to find a way to add magnets or change filament. With Luban I can't do that on a layer based trigger, only by following the print and manually pause. I think Cura has this features... I'm investigating.

tyeth, 2021-10-04:
> Hi, I'm also curious about this. I have a SM2-A350 and I need to find a way to add magnets or change filament. With Luban I can't do that on a layer based trigger, only by following the print and manually pause. I think Cura has this features... I'm investigating.

Closest I came to that was trying to trap a fabric screen in one of the layers of a print (think mosquito net for power supply), and ended up adding a 30minute pause/delay to the print at the crucial moment. It involves checking on the current layer and estimating when the pause layer might occur, or just go and check regularly and if it's not moving then stick in the magnets :)

uniederer, 2022-01-26:
Seems as if this feature request aims into a similiar direction: #1360

tcmonster, 2024-09-27:
Is someone solving this issue? This should be a very valuable function.

SanielDous, 2024-09-27:
I solve this by manually adding the `M600` command to the gcode file. Find more information about the command here:
https://wiki.snapmaker.com/en/Snapmaker_Luban/manual/2_supported_gcode_references#m600-modified-trigger-filament-runout-event
It is important to consider the notes of that command.