Snapmaker/Luban #1040: Bug: Package Path Name contains a space

ITmaze, 2021-06-27:
The directory name for a Linux installation is "/opt/Snapmaker Luban/". This causes electron to crash with a core dump.

I'm seeing this in the current version, 3.15.1.

Note that I have been attempting to use Luban inside Docker for several versions and only managed to find this issue with this particular version, but I suspect that this has been the same problem I've previously experienced.

Note: I'll also draw your attention to a missing dependency (libasound2), but I'll create a separate issue for that.

Dockerfile:
<pre><code>
FROM debian:stable-slim
ADD https://github.com/Snapmaker/Luban/releases/download/v3.15.1/snapmaker-luban-3.15.1-linux-amd64.deb /snapmaker-luban.deb

ARG DEBIAN_FRONTEND=noninteractive

RUN apt-get update
RUN apt-get -y upgrade

RUN apt -y install /snapmaker-luban.deb

RUN useradd -ms /bin/bash docker

RUN mv /opt/Snapmaker\ Luban /opt/Snapmaker_Luban
RUN rm /usr/bin/snapmaker-luban
RUN ln -s /opt/Snapmaker_Luban/snapmaker-luban /usr/bin/

RUN apt -y install libasound2

USER docker

ENTRYPOINT ["/usr/bin/snapmaker-luban"]
CMD ["--help"]
</code></pre>

ITmaze, 2021-06-27:
See Issue #1041 for the missing dependency.

parachvte, 2021-06-28:
`mv /opt/Snapmaker\ Luban /opt/Snapmaker_Luban`

Is the space you mentioned the one goes after "Snapmaker\ Luban"?
I don't know for sure what caused the trailing space in Linux.

ITmaze, 2021-06-28:
No, it's the space between "Snapmaker" and "Luban".

parachvte, 2021-06-28:
So better we change it back to "snapmaker-luban"?

For Windows and macOS packages, it was changed to "Snapmaker Luban" since we moved CI to Github Actions (https://github.com/Snapmaker/Luban/pull/827).

Looks like we didn't make changes for Linux build for a while..

ITmaze, 2021-06-28:
`snapmaker-luban` seems like a reasonable name. I've just rebuilt my Dockerfile and that works fine.

ITmaze, 2021-06-28:
FYI as an aside, I'm working on making a Dockerfile that can run Luban headless, so it can sit inside a container listening to a port and get accessed with a web-browser.

parachvte, 2021-08-09:
From Electron [docs](https://www.electron.build/configuration/configuration), changing the `productName` field will also change the installation location of software. I've verified it on Debian.

But changing productName may also change the software name on macOS and Windows, which is unexpected. We want to remain the software name as "Snapmaker Luban".

Another approach will be modify `package.json` when building Linux packages. It brings some complexity so we'll defer the change. @ITmaze

closed 2021-08-09 by parachvte