Snapmaker/Luban #1093: Purifier is not exist... when connected via usb?

systemsmeister, 2021-07-30:
When connected to usb, this message continually comes up in Luban console (most up to date firmware)... need to disconnect it to stop the message in the console.

others < Module 0x003B1B66: v1.10.3
others < Module 0x003B1F88: v1.10.3
others < BT version: V1.5.1.0-19122015
others < Purifier is not exist.
snapmaker-luban 3.15.1
Connected to COM4
G54
ok
others < Purifier is not exist.
others < Purifier is not exist.

gotcottonmouth, 2021-08-03:
I actually just got this same error, anyone have any feedback? 

Im troubleshooting bed leveling issues and running the commands via terminal and just came across this so I'm not sure when the issue started or if its related to my bed leveling issues.

Just in case its related:
Up to this point, I've always used the on-screen features of the SN2
M420 V reports +641.000 for all points sampled via the on-screen bed leveling process.
When running G1029 P11, the leveling distance sensor is recording what appears to be accurate data.
G1029 S effectively records the data
M420 V now reports correct data per the terminal process of bed leveling

Next:
Starting a large print that covers about 75% of the total surface of the print bed.

sumibin, 2021-08-04:
We'll release a new version to fix it as soon as possible.

gotcottonmouth, 2021-08-04:
Any relationship to the issues I've reported or is it just a back-end problem?

sumibin, 2021-08-04:
> Any relationship to the issues I've reported or is it just a back-end problem?

Just because the console doesn't filter out invalid information.

gotcottonmouth, 2021-08-04:
Okay. Thanks for the prompt reply, I will continue to research my problem.

parachvte, 2021-08-09:
Please checkout Luban v3.15.2 (https://github.com/Snapmaker/Luban/releases/tag/v3.15.2).

The issue should be fixed.

closed 2021-08-09 by parachvte