Snapmaker/Luban #1203: Bug: timecalculations for lasercutting/laseretching

lichtwellenreiter, 2021-09-18:
# 🐞 bug report

### Affected Version(s)
The issue is caused in Luban 4.0.2

### Is this a regression? (optional)
The issues did not happen in prior versions to 4.x.x

### To Reproduce
Steps to reproduce the behavior:
1. Create a design
2. load into Luban Laser module
3. generate .nc
4. see calculated time

Expected behavior
The time calculations should be preciser. Calculated time by Luban was 1h 42 min. Effective laser etching took 4h 22min. Screenshot from Luban:
[image]

## 🌍  Your Environment
- Operating System: Windows 10 x64, macOS Big Sur
- Printer: A250

F1nnM, 2022-01-11:
I'd like to add that there are also cases, where luban seriously overestimates the time.
With fill:dots it calculates about three times as much time as it actually takes, in my cases at least.

sumibin, 2023-03-21:
We have optimized the motion part of the firmware, as well as the time firmware of Luban at the same time. This issue will be closed.

closed 2023-03-21 by sumibin