Snapmaker/Luban #1205: Bug: Multi-toolpath only acknowledges laser power setting of first tool-path

tapeboy7, 2021-09-20:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by versions 
4.0.0
4.0.1
4.0.3

### Is this a regression?

 Yes, the previous version in which this bug was not present was: 3.15.2

### To Reproduce
Steps to reproduce the behavior:
1. Start a new laser cnc project
2. Add multiple objects or shapes in the editor
3. Create separate toolpath for each object
4. Make sure each toolpath (process) has a different laser power setting (e.g. 10% and 100%)
5. Click _Generate G-code and Preview_  -> _Export_ -> _Load G-code to Workspace_
6. Proceed to run code (either from Luban or transfer to Snapmaker and run)
7. See problem: Only the laser power setting of the first toolpath is acknowledged. 
 **The G-code compiler sets laser power of the first toolpath as the global laser power  in the header of the G-code file**
The G-code below the header section correctly has the power levels set for each operation.

Error in G-code
<pre><code>

;Header Start
;header_type: laser
;renderMethod: line
;file_total_lines: 120
;estimated_time(s): 67.958
;is_rotate: false
;diameter: 40
;max_x(mm): 5
;max_y(mm): 5
;max_z(mm): 0
;max_b(mm): 0
;min_x(mm): -5
;min_y(mm): -5
;min_b(mm): 0
;min_z(mm): 0
;work_speed(mm/minute): 140
;jog_speed(mm/minute): 3000
**;power(%): 10**
;thumbnail: data:image/png;base64,<image data string>
;Header End

; Laser multi-pass, pass 1 with Z = 0
; G-code for laser engraving
; Generated by Snapmaker Luban
; G-code START <<<
G90
G21
M106 P0 S255
G0 F3000
G1 F140
G0 X-5.00 Y5.00
**M3 P10 S25**
M3
G1 X5.00 Y5.00
G1 X5.00 Y-5.00
G1 X-5.00 Y-5.00
G1 X-5.00 Y5.00
M5
G0 X0.00 Y0.00
M107 P0
; G-code END <<<
; Laser multi-pass, pass 2 with Z = -0.6
; Laser multi-pass: dropping z
G91
G0 Z-0.6 F150
G90
; G-code for laser engraving
; Generated by Snapmaker Luban
; G-code START <<<
G90
G21
M106 P0 S255
G0 F3000
G1 F140
G0 X-5.00 Y5.00
**M3 P10 S25**
M3
G1 X5.00 Y5.00
G1 X5.00 Y-5.00
G1 X-5.00 Y-5.00
G1 X-5.00 Y5.00
M5
G0 X0.00 Y0.00
M107 P0
; G-code END <<<
G91
G0 Z0.6 F150
G90

;Header Start
;renderMethod: line
;Header End

; Laser multi-pass, pass 1 with Z = 0
; G-code for laser engraving
; Generated by Snapmaker Luban
; G-code START <<<
G90
G21
M106 P0 S255
G0 F3000
G1 F140
G0 X-2.50 Y2.50
**M3 P100 S255**
M3
G1 X2.50 Y2.50
G1 X2.50 Y-2.50
G1 X-2.50 Y-2.50
G1 X-2.50 Y2.50
M5
G0 X0.00 Y0.00
M107 P0
; G-code END <<<
; Laser multi-pass, pass 2 with Z = -0.6
; Laser multi-pass: dropping z
G91
G0 Z-0.6 F150
G90
; G-code for laser engraving
; Generated by Snapmaker Luban
; G-code START <<<
G90
G21
M106 P0 S255
G0 F3000
G1 F140
G0 X-2.50 Y2.50
**M3 P100 S255**
M3
G1 X2.50 Y2.50
G1 X2.50 Y-2.50
G1 X-2.50 Y-2.50
G1 X-2.50 Y2.50
M5
G0 X0.00 Y0.00
M107 P0
; G-code END <<<
G91
G0 Z0.6 F150
G90
</code></pre>

Expected behavior

The laser power setting in the header section should be omitted:

<pre><code>
;power(%): 10
</code></pre>

The Snapmaker should use the laser power setting from the _M3_ line in the G-code:
<pre><code>
M3 P10 S25
;M3 [P<power in percentage>] [S<power in strength 0-255>]
</code></pre>

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: macOS Big Sur
- Printer: Snapmaker A350

</code></pre>

FriedSemiconductors, 2021-09-20:
I had a similar/related bug with Luban 4.0.1, but it was not the power of the first toolpath.
I wanted to laser engrave a greyscale image at 20% power and the second toolpath should vector cut out the picture at 100% power

The greyscale image started burning at 100% power, gladly I noticed right away and aborted. This could be really dangerous. 🔥

I have not yet tried with Luban 4.0.2 or 4.0.3

sumibin, 2021-09-22:
`;power(%): 10`
It's code will not be used for actual processing.This is only used when G-code information is displayed on the screen.

haringsrob, 2021-11-17:
Is this related to: https://forum.snapmaker.com/t/luban-4-0-0-wont-turn-the-laser-to-100/20494

closed 2023-03-21 by sumibin