Snapmaker/Luban #1206: Bug: Dodgy GCode in 3d printing

tyeth, 2021-09-21:
# 🐞 bug report

### Affected Version(s)
4.0.3

### To Reproduce
Steps to reproduce the behavior:
1. Set machine to A350, do something close and reopen luban. Set machine to Snapmaker original. do something, close and reload luban.
2. Click on "Fabric" / chain-mail 3dp sample in Case Study panel of Luban 
3. Accept defaults of Normal quality, PLA, and generate and export gcode.
4. Take via USB to snapmaker, or dont bother just read gcode and **spot the issue, temp is 215c**, but clearly states in PLA material settings 205.
5. Also notice seriously long base64 thumbnail included as comment in gcode, is this actually used because my snapmaker2 has never loaded one preview so far just white box and the snapmaker1 I just got has no such feature as preview (nor bed/nozzle temp adjust mid print but old firmware so will update before bug report).

Expected behavior
Material and Quality settings should be adhered to.
Fabric Gcode generated by luban 4.0.3 linked here:
https://gist.github.com/tyeth/8a916517e061cddc9b6a93a1df18eeee

## 🌍  Your Environment

**Platform:**
- Operating System: Windows 10 x64
- Printer: Snapmaker Original first print (had luban setup for my snapmaker A350 hence the reproduction steps)

sumibin, 2021-09-22:
1. "Fabric" Temp : We confirmed that the temperature of the case was 205 ℃(PLA). Maybe you changed the parameters in PLA?
[image]

2. Thumbnail : This feature can cause a serious exception, so we went offline it at 4.0.**We'll fix it in 4.1.**

tyeth, 2021-09-22:
No, it's clearer now, I reload the app and the default selected PLA isn't respected. If I then select ABS it generates the correct GCODE, if I then select PLA it correctly uses 205, if I then shut Luban and reload it, selecting the fabric example again then generate gcode it is back at 215c. If I select the materials dropdown and reclick on PLA it still is 215.
 I have to select a different material and then back to PLA to get it to respect PLA after first loading the program and example print (fabric)

tyeth, 2021-09-22:
I can provide gcode (and a video) if really necessary

tyeth, 2021-10-13:
This sounds like maybe I once set PLA to 215, and due to you implementing remembering last user value then when luban loads it uses an old user value and confusingly lists that PLA is selected (which is true but the temp value doesn't match the PLA temp value due to being a remembered and unforgettable user set value) meaning the user can't tell if the PLA profile is being used or a modified set of values. 
Related #1180

sumibin, 2023-03-21:
This problem should have been solved by now

closed 2023-03-21 by sumibin