Snapmaker/Luban #1214: CNC module crashes under Ubuntu Linux

ozzyman, 2021-09-27:
The CNC module in luban V4.0.3 (also V4.0.0) fails to create toolpaths on Ubuntu Linux. 

When I click "create toolpath" the screen goes blank with a menu bar at the top. Nothing else happens. If I select "reload" from the "Window" menu, Luban crashes and restarts. 

Bug is new in version 4.0.x  Toolpath creation works as expected in version 3.15

[image]
[image]
[image]
[image]
.2

raivonigul, 2021-10-04:
I have same issue. Ealier version of Luban works properly. Current version works propely on Windows. In Linux the toolpath dialog are created - wisible on task bar as double Icon but not visible on workspace

ozzyman, 2021-10-06:
I don't even see the second window under Ubuntu 20.04

deimi, 2021-12-08:
It seems that it works with v4.1.0-beta.1

jane-rose, 2021-12-14:
Yes, it is fixed in v4.1.x version. But the beta version is unstable, please update to the official version 4.1.1 or later.

> It seems that it works with v4.1.0-beta.1

closed 2023-03-21 by sumibin