Snapmaker/Luban #1237: Feature Request: Option to disable fan for initial layers when 3D printing

wdecook, 2021-10-20:
Sometimes, it's just impossible to get that first layer down without it warping and pulling up off the board.

In this case, it would be handy to be able to disable the cooling fans for the initial layer(s) in order to slow the cooling of the filament as it comes out of the hot end and decrease the likelihood of warping.

pavolzibrita, 2022-09-03:
Some filaments suggest that, e.g. https://www.amazon.ca/gp/product/B09DKP4TLW/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1 

```
【Note - turn off the fan for the first layer for improved bed adhesion】
```

sumibin, 2023-03-21:
This feature has been released.

closed 2023-03-21 by sumibin

parachvte, 2023-03-22:
[image]

By using "All" filter, you can see the "Regular Fan Speed at Layer = 2".

By default, the fan should be off for first few layers (Initial Layer Optimization -> Initial Fan Speed = 0%), and starts to work at regular speed from layer 2.