Snapmaker/Luban #1286: Feature Request: WYSIWYG in laser/cnc

jenkins007, 2021-11-13:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.

### Describe the solution you'd like?

jenkins007, 2021-11-13:
Trying to etched pictures using A350 and Luban (4.0.3). Imported a .jpg picture. the software appears not to use WYSIWYG. For example, when the same picture is small, it looks on the screen extremely dark. When zooming in, the picture becomes lighter and more readable. See the attached screen shots for the example. Please note that no changes were made other than to use the mouse to zoom in and zoom out. When looking at the picture when it is small, the picture is so dark it is unrecognizable. 

[image]
[image]
[image]

jenkins007, 2021-11-22:
This is using the Mac version.

kalmdown, 2021-12-18:
I would call this a defect. Images should maintain their tonal values regardless of scale.