Snapmaker/Luban #1301: Bug: 4-axis laser (and possibly CNC) has confusing x-axis origin rendering

maihde, 2021-11-22:
# 🐞 bug report

### Affected Version(s)
- 4.0.3
- HEAD

### To Reproduce

1. Start any 4-axis Laser project
2. Observe that the '0' for the x-axis is rendered differently than other x-axis labels; it is not outside of the work area and it is not centered on zero.

Expected behavior

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [Windows 10 x64, macOS Big Sur, Ubuntu 20.04, etc]
- Printer: [Snapmaker Original, Snapmaker A150, etc]

</code></pre>

maihde, 2021-11-23:
I have played around with the rendering in `PrintableArea.js` I personally find this to be the most intuitive display for a user, but understand that it is somewhat subjective.  The zero renders aligned with the x-axis (since the y-axis origin is very obvious in it's location) and the x-axis is render slightly thicker (I also consider doing darker, a different color, or dashed).

[image]

As a side-effect of the change, 2D projects will have their grid rendered like this:

[image]

Which I find to be a small improvement as well, but if there is a desire to not change the 2D rendering in any manner I understand.

I'll wait to see if there is any feedback prior to make a PR.

sumibin, 2021-11-24:
> 

I think this is a good change. The bold axis can improve both types of projects. The result of zero-aligned to X-axis and Y-axis is not very different for 2D projects, but it can improve four-axis projects.

closed 2023-03-21 by sumibin